jdunmire/kicad-ESP8266 #8: kicad doesn't import the module

podema, 2015-10-22:
In CvPcb or PcbNew when opening the libraries dialog i can't import the footprints, since kicad requires a .mod.
I tried renaming the files from _mod to .mod and add the folder as a search path. None worked

nullydragon, 2015-10-22:
I suggest getting a new version of kicad. kicad_mod's work fine in version (2015-10-05 BZR 6247)-product release build

jdunmire, 2015-10-22:
@podema, what version of kicad are you running? Can you provide more specifics about the operations you've tried?
I have had no problem opening the libraries across several versions of kicad. The version I am running is 0.201510211816+6273~30~ubuntu15.04.1-product,release build

podema, 2015-10-23:
i'm running build 25-oct-2014 bzr 4029 installed through the package manager (ubuntu 15.04) package: kicad 0.20141025+bzr4029-2.

I tried to import it through preferencies > libraries and then add, the file browser automatic filters .mod so i can't select anything, just to try i renamed the filed without succeed.

I see that i'm running an old build, and this is causing some sort of uncompatibilitie. Since you are running ubuntu which package have you installed, from which repo?

Thanks in advance

podema, 2015-10-23:
I downloaded the 4.0 rc from the kicad page and it is now working. 
My apologies, i'm new with kicad and pcb design, i did not expect version issues

closed 2015-12-09 by jdunmire