jdunmire/kicad-ESP8266 #28: Wrong 3D Model for ESP-12E_SMD.kicad_mod

alexwhittemore, 2017-08-09:
It looks like the 3D model associated in this footprint is ESP-12.wrl when it should be ESP-12E.wrl.

Caveat: I'm an extremely novice kicad user JUST jumping in to 3D and have no idea what I'm doing. Also, on that subject: this library makes use of "${ESPLIB}" for defining the path to the 3D models. How does that work? Where does it get defined? I've installed this library with a "git clone" to Users/Me/KiCad/Libraries/kicad-ESP8266 (this is windows), and none of the 3D models get found until I manually add them at the footprint level with the absolute path... what am I doing wrong? I'd love a pointer.

jdunmire, 2017-08-10:
I tend to agree, but I won't be able to confirm it for awhile. Meantime if you can test out the ESP-12E.wrl, make the change and submit a pull request I would be happy to merge it.

You might also try contacting Stephen Legge here on GitHub. He submitted the patch to add the ESP-12E_SMD.kicad_mod.

WereCatf, 2017-09-13:
@alexwhittemore ; Say, if your project was in C:\Users\alex\fancyproject, you'd place the ESP8266.3dshapes-folder in C:\Users\alex\fancyproject\ESP8266.3dshapes. Now, open up footprint-editor, then select Preferences -> Configure Paths, then click Add and type ESPLIB on the left box and the full path to your project on the right one, ie. C:\Users\alex\fancyproject -- note that you do not add the full path to the 3dshapes-folder, just the upper-level folder.

@jdunmire I came here to say the same thing as Alex there, it seems the ESP-12E SMD-footprint has the wrong model associated with it. ESP-12E.wrl seems to be the correct one to use.