jdunmire/kicad-ESP8266 #29: Footprints don't show up in browser

alexwhittemore, 2017-08-09:
Open project, click "PCB Footprint Editor." Click "Select active library" then find "ESP8266" Click "Load footprint from library" then "List all" This results in the message "No footprints could be read from library file(s): in any of the library search paths. Verify your system is configured properly so the footprint libraries can be found." - Everything appears to be installed correctly, and I can use the components in a design, so what have I done wrong? Additionally, in the Library Browser (click "Open footprint viewer") no footprints show up under the ESP8266 library.

This is probably my own fault given that I don't know what I'm doing, but after a few minutes of compare and contrast, other libraries work. I've got this library installed at "Users\me\KiCad\Libraries" as per readme.md (also that spot makes sense to me).

jdunmire, 2017-08-10:
It looks like you've chosen a reasonable path for the library, though it appears you are using Windows and I don't have any experience with it (Linux only for years).

My footprint editor does not have a 'Find' command. The closest I see is an icon with a magnifying glass over some books that opens the 'Footprint Viewer'. When I click on that, it is the 'Library Browser' that comes up. When I select ESP8266 in the left column a list of foot prints appears in the middle column. Selecting one of those shows the footprint.
[image]
Hope that helps.

robin48gx, 2018-07-31:
I see no footprints at all in the library browser.