jdunmire/kicad-ESP8266 #32: KiCad library integration

antoniovazquezblanco, 2017-10-19:
I wonder why haven't you tried to push your symbols, models and footprints to KiCad.

I have interest in using ESP8266 but have not yet designed any circuit with it. I would love to see your models in the KiCad library for easiness of use.

I hope you find the idea appealing!

Nice work ;)

werty37, 2017-10-21:
Hi

I have the same request. Also i was trying to import the footprint using "Add footprints library wizard" and somehow it is not showing up. The URL i am using is 👍 
https://github.com/jdunmire/kicad-ESP8266/

Cheers
S

fandres, 2018-02-14:
I have the same request.
https://github.com/KiCad/kicad-symbols
https://github.com/KiCad/kicad-footprints

igagis, 2018-07-24:
In order to make this library visible by `library wizard` the repo name should be `ESP8266.pretty` and `kicad_mod` files should be in the root of the repo. Then wizard will be able to detect this as a library.

mbbackus, 2018-10-20:
I tried using your esp-01, but I get the following error message after following your instructions: No footprint defined for component.