jdunmire/kicad-ESP8266 #41: Feature Request: KiCad 5 Version

mjf55, 2018-05-21:
This is a great library, but unfortunately, I could not use them in KiCad  5.0 ( nightly build, Application: kicad
Version: (5.0.0-rc2-dev-744-gdfcdfe91f), release build)).  When I tried (in eeschema) , I got an error saying it could not read the lib.  If you want more specific information, I will re-install and try again. just let me know what you want to see.

With the coming release of KiCad 5.0, any chance of updating them for this release?

jdunmire, 2018-05-24:
Thank you for the support. I would like to get the library working on KiCad 5.0, but I don't have much time to investigate the problem right now. I would hope it is something simple.

If there are any suggestions in the error message about what needs fixed, please post it.

mjf55, 2018-05-24:
Sorry about this.  I just re-tried using your library and all the components loaded fine.  No errors.  I must of messed up my original preferences settings.  Closing as keyboard (the person using ) error.   Thanks for a nice library.

closed 2018-05-24 by mjf55