jdunmire/kicad-ESP8266 #44: ESP822-12E: Wrong pinout in schematics

tobiassoltermann, 2018-10-27:
Great library, thanks for the amazing work.
There's just one thing I've discovered. The PIN number order from 9 to 22 in the schematics does not correctly match to the one in the footprints.

The datasheet tells that pin ordering is the form of an U. It starts on top left on 1 then goes down the left edge (that part is okay). Then PIN 9 to 14 continue on the bottom edge and then go up again on the right edge from 15 to 22. 

Here's the schematics (label ordering is correct, pin numbers are not)
[image]

Here's the footprint (numbering is correct, labels are not)
[image]

Pinout as it should be:
[image]

Thanks for looking into it.

phschoen, 2019-05-29:
hi Jerry,
this is an odd approach. Then the compatible part should have separate part. The normal esp-12 and esp12e should be like they are in real. And i mean if you would break existing designs this is partial true since you can always use git sub-module with the exact version of the lib you where doing your design.
Otherwise this flaw can never be corrected and even new one would have it.

WBR Philipp

jdunmire, 2019-05-29:
Hi Philipp,
I understand your position on this approach. It was very useful at the time (which actually pre-dates this thread) and it remains internally consistent.

An approach to clarifying the problem might be to create a 'vendor numbering' branch going forward, or at the risk of accidental updates to old designs (yes, I know sub-modules could be used to deal with the problem), maybe a 'compatibility' or 'historic' branch.

What ever approach is pursed, I no longer have the time to support this library. I sincerely hope it has been useful to others and that someone has the time and interest to take it up.

Regards,
...jerry